sonydevworld/spresense #131: Not all submodules are out of 'Detached HEAD' state

rajames, 2021-03-22:
## Description of the issue:

Following the instructions in the README to get the submodules out of detached
state results in that `sdk/apps` submodule is still in detached state.

## How to reproduce:

``` sh
$ git clone --recursive https://github.com/sonydevworld/spresense.git
$ git submodule foreach git checkout master
```

As result:

``` sh
$ git submodule foreach git checkout master
Entering 'externals/nnablart/nnabla-c-runtime'
Already on 'master'
Your branch is up to date with 'origin/master'.
Entering 'nuttx'
Already on 'master'
Your branch is up to date with 'origin/master'.
Entering 'sdk/apps'
error: pathspec 'master' did not match any file(s) known to git
fatal: run_command returned non-zero status for sdk/apps

```

## The problem:

`sdk/apps` submodule has no branch `master` rather `new-master`.

## Describe your hardware setup:

## Software version used:

kamtom480, 2021-03-23:
Hi @rajames 

Please skip this command and try again:
`$ git submodule foreach git checkout master`

In the new SDK running this command is not needed. I will update the README. Thanks for finding this.

rajames, 2021-03-23:
@kamtom480 you're welcome

closed 2021-04-21 by kamtom480