sonydevworld/spresense #132: Issues configuring or building an example with pre-compiled docker image

rajames, 2021-03-22:
## Description of the issue:

When using the prebuilt docker image I'm unable to configure and/or build the
`hello` example after following instructions in README.

# How to reproduce:

``` sh
$ git clone --recursive https://github.com/sonydevworld/spresense.git
$ git submodule foreach git checkout master
```

Then using the docker image:

``` sh

$ source spresense_env.sh 
This script has to be sourced.
. ./spresense_env.sh or source spresense_env.sh
 
Using default tag: latest
latest: Pulling from devworldsony/spresense-sdk-env
Digest: sha256:3a2f02cf9e0a1063585db66ddca43d5633dd65545ea9d00dd88983de39ca24d6
Status: Image is up to date for devworldsony/spresense-sdk-env:latest
docker.io/devworldsony/spresense-sdk-env:latest
After this script executed successfully:
Usage: spresense [command]
This will run command in the Spresense SDK docker environment.
SpresenseSDK:$ spresense sh -c "cd /spresense/sdk && tools/config.py examples/hello"
make[1]: *** [include/arch/board] Error 1
make: *** [olddefconfig] Error 2
Post process failed. 512
Try 'make distclean' first.
SpresenseSDK:➜  spresense git:(docker-readme) ✗ 

```

## Describe your hardware setup:

## Software version used:

On master (commit: `6cb78083`)

rajames, 2021-03-22:
It appears that the following:

``` sh
git submodule foreach git checkout master
```

updates the submodules:

``` diff
modified   externals/nnablart/nnabla-c-runtime
@@ -1 +1 @@
-Subproject commit c58c11ddd182e8ed04eb931bbf2a82f190874c9d
+Subproject commit 57dd88dde097a51cd1864810814afc31eef9de5a
modified   nuttx
@@ -1 +1 @@
-Subproject commit 7686b71a7cfe17bf8f3fb1be7749594278b0c808
+Subproject commit f28c114ccdd5c58b2a372634d71440c676bd2fae
```

reverting the above changes seems to give better results (or fixed the problem)

kamtom480, 2021-03-23:
Hi @rajames 

Does this mean the problem has been resolved and we can close this issue?

rajames, 2021-03-23:
Hi @kamtom480 the problem is not resolved since following the README instructions I get the above described issue.

kamtom480, 2021-03-23:
@rajames 

Thank you for your confirmation. I created a PR #133 for this. When the #133  is merged I will close this issue.

fxmarty, 2021-04-16:
Hi, in which way this git issue is related with `make olddefconfig` failing?

rajames, 2021-04-21:
@fxmarty Sorry, I'm not familiar with issue related with `make olddefconfig`

kamtom480, 2021-04-21:
@fxmarty This issue should not be related to `make olddefconfig` failing. Do you have problems with `make olddefconfig`?

fxmarty, 2021-04-28:
Sorry to pollute this issue then, I will post this single message.

Yes, spresense SDK's `tools/config.py` (e.g. called as `./tools/config.py examples/hello`) would fail for me in `do_olddefconfig()`, when we try to perform `make olddefconfig`. Despite doing `make distclean` as suggested the error persists.

The actual root problem is the following, when executing `make olddefconfig`:
```
make[2]: Leaving directory '/home/felix/spresense/sdk/apps'
../sdk/apps/Kconfig:10: can't open file "/home/felix/spresense/sdk/apps/examples/Kconfig"
```

I wonder why in my case this Kconfig file is missing. I found a manual solution which at least does not show errors, which is to delete `sdk/Kconfig` and ``sdk/apps/Kconfig` files. Deleting ``nuttx/.config` is an alternative which also works.

After deleting such files, `tools/config.py` do not complain anymore.

kamtom480, 2021-04-29:
@fxmarty Could you please send me step by step what you do from the moment the repository is cloned until an error occurs? 

What OS are you using?