sonydevworld/spresense #158: Can't use <string> in C++

automatiek, 2021-04-22:
Hi!

In order to do C++ string manipulations we are using `<string>` . But including this file results in following errors:

```
arm-none-eabi-ld: /home/arjan/Documents/EdgeImpulse/Firmware/sony/spresense/nuttx/staging/libapps.a(helloxx_main.o): in function `void std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_construct<char const*>(char const*, char const*, std::forward_iterator_tag)':
/usr/include/newlib/c++/9.2.1/bits/basic_string.tcc:225: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_S_copy_chars(char*, char const*, char const*)'
arm-none-eabi-ld: /home/arjan/Documents/EdgeImpulse/Firmware/sony/spresense/nuttx/staging/libapps.a(helloxx_main.o): in function `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::~basic_string()':
/usr/include/newlib/c++/9.2.1/bits/basic_string.h:658: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_dispose()'
```

Meaning that the std libs don't support the string. Could you provide support for this?

kamtom480, 2021-04-23:
Hi @automatiek 

I tried to add `#include <string>` to helloxx example and then I ran:
```
$ ./tools/config.py examples/helloxx
$ make
```
It worked for me.

Could you please try to use the arm-none-eabi version 7.3.1?

automatiek, 2021-04-23:
HI @kamtom480,

Thanks for checking. Could you then create instance of a string like:
```
std::string hello = "hello, I'm a string";
```

and check if building the code still works.

Yes, I'm running 7.3.1

kamtom480, 2021-04-26:
Hi @automatiek 

The issue is that **NuttX** does not support **standard C++ libraries** by default. Fortunately, there is support for an external module that adds this. This is ***uClibc++*** which you can install using the instructions from here: https://bitbucket.org/nuttx/uclibc/src/master/README.txt. You should modify this file: `sdk/tools/scripts/Make.defs`.

I tried it and was able to build the helloxx application with your suggested changes. I hope it will help.

automatiek, 2021-04-26:
Hi @kamtom480,

Great that you've got it working. I'm stuck here:

```
arm-none-eabi-ld: error: /app/spresense/nuttx/nuttx uses VFP register arguments, /root/spresenseenv/usr/bin/../lib/gcc/arm-none-eabi/7.3.1/../../../../arm-none-eabi/lib/libsupc++.a(vterminate.o) does not
arm-none-eabi-ld: failed to merge target specific data of file /root/spresenseenv/usr/bin/../lib/gcc/arm-none-eabi/7.3.1/../../../../arm-none-eabi/lib/libsupc++.a(vterminate.o)
arm-none-eabi-ld: error: /app/spresense/nuttx/nuttx uses VFP register arguments, /root/spresenseenv/usr/bin/../lib/gcc/arm-none-eabi/7.3.1/../../../../arm-none-eabi/lib/libsupc++.a(cp-demangle.o) does not
arm-none-eabi-ld: failed to merge target specific data of file /root/spresenseenv/usr/bin/../lib/gcc/arm-none-eabi/7.3.1/../../../../arm-none-eabi/lib/libsupc++.a(cp-demangle.o)
arm-none-eabi-ld: error: /app/spresense/nuttx/nuttx uses VFP register arguments, /root/spresenseenv/usr/bin/../lib/gcc/arm-none-eabi/7.3.1/../../../../arm-none-eabi/lib/libsupc++.a(eh_type.o) does not
arm-none-eabi-ld: failed to merge target specific data of file /root/spresenseenv/usr/bin/../lib/gcc/arm-none-eabi/7.3.1/../../../../arm-none-eabi/lib/libsupc++.a(eh_type.o)
/app/spresense/nuttx/staging/libapps.a(helloxx_main.o): In function `helloxx_main':
/app/spresense/sdk/apps/examples/helloxx/helloxx_main.cxx:143: undefined reference to `std::basic_string<char, std::char_traits<char>, std::allocator<char> >::basic_string(char const*, std::allocator<char> const&)'
/app/spresense/sdk/apps/examples/helloxx/helloxx_main.cxx:143: undefined reference to `std::basic_string<char, std::char_traits<char>, std::allocator<char> >::~basic_string()'
/root/spresenseenv/usr/bin/../lib/gcc/arm-none-eabi/7.3.1/../../../../arm-none-eabi/lib/libsupc++.a(vterminate.o): In function `__gnu_cxx::__verbose_terminate_handler()':
vterminate.cc:(.text._ZN9__gnu_cxx27__verbose_terminate_handlerEv+0x184): undefined reference to `_impure_ptr'
```
I saw a workaround for the `vterminate.o` already. But not yet for the VFP error. Something with the `mfloat-abi` settings probably. Any clues on where and how to set those?

kamtom480, 2021-04-26:
Hi @automatiek 

Please change this line: 
```
LIBSUPXX = ${shell $(CC) --print-file-name=libsupc++.a}
```
to:
```
LIBSUPXX = ${shell $(CC) $(CFLAGS) --print-file-name=libsupc++.a}
```

automatiek, 2021-04-28:
Great this works! Thanks for you help,

closed 2021-04-28 by automatiek