sonydevworld/spresense #200: Importing inttypes.h fails due to misplaced wchar_t definition

guberti, 2021-07-05:
### Description of the issue:
Including `inttypes.h` in a `.c` file fails.

### How to reproduce:
The issue should be easy to reproduce by just creating a `importer.c` file with a single line:
```
#include <inttypes.h>
```
Compiling that file will produce the following errors:
```
In file included from /home/guberti/Desktop/demonstrate_inttypes_bug/importer.c:1:0:
/home/guberti/.arduino15/packages/SPRESENSE/tools/spresense-sdk/2.2.1/spresense/release/nuttx/include/inttypes.h:185:31: error: unknown type name 'wchar_t'
 intmax_t  wcstoimax(FAR const wchar_t *nptr, FAR wchar_t **endptr, int base);
                               ^
/home/guberti/.arduino15/packages/SPRESENSE/tools/spresense-sdk/2.2.1/spresense/release/nuttx/include/inttypes.h:185:50: error: unknown type name 'wchar_t'
 intmax_t  wcstoimax(FAR const wchar_t *nptr, FAR wchar_t **endptr, int base);
                                                  ^
/home/guberti/.arduino15/packages/SPRESENSE/tools/spresense-sdk/2.2.1/spresense/release/nuttx/include/inttypes.h:186:31: error: unknown type name 'wchar_t'
 uintmax_t wcstoumax(FAR const wchar_t *nptr, FAR wchar_t **endptr, int base);
                               ^
/home/guberti/.arduino15/packages/SPRESENSE/tools/spresense-sdk/2.2.1/spresense/release/nuttx/include/inttypes.h:186:50: error: unknown type name 'wchar_t'
 uintmax_t wcstoumax(FAR const wchar_t *nptr, FAR wchar_t **endptr, int base);
                                                  ^
```
You can also [use this Arduino script](https://gist.github.com/guberti/fcef3acdb3894533b42c64becc46b478) to reproduce the issue.

SPRESENSE, 2021-07-06:
@guberti 
Thank you for reporting the issue.

This has been already resolved in the following commit on the `develop` branch, and will be released in the next official version.
https://github.com/sonydevworld/spresense-nuttx/commit/e99e185712bc19b969dcc0f9db5557234c7be25e#diff-18fc24aec5247a51ba32867677c04c53a35eda8f3407f7d94c55318867990088

Thanks.

SPRESENSE, 2021-09-03:
SDK v2.3 has been released and this issue has been resolved.

closed 2021-09-03 by SPRESENSE