sonydevworld/spresense #346: The revision nuttx-tools written at README.md fails "tools/config.py examples/hello"

sntulix, 2021-12-29:
description:
nuttx-tools with the rev in README.md fails config.py examples.
```
spresense/sdk$ tools/config.py examples/hello
arch/arm/src/imxrt/Kconfig:1236: syntax error
arch/arm/src/imxrt/Kconfig:1235: invalid option
arch/arm/src/stm32l4/Kconfig:5236: syntax error
arch/arm/src/stm32l4/Kconfig:5185: missing end statement for this entry
arch/arm/src/stm32l4/Kconfig:5053: missing end statement for this entry
arch/arm/src/stm32l4/Kconfig:8: missing end statement for this entry
arch/arm/Kconfig:1044: missing end statement for this entry
...
```

My environment below.
```
$cat /etc/lsb-release 
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=16.04
DISTRIB_CODENAME=xenial
DISTRIB_DESCRIPTION="Ubuntu 16.04.2 LTS"
```
```
spresense$ git branch
* master
```
```
spresense/tools/kconfig-frontends$ git branch
* (HEAD detached at 891510d)
  master
```

Fails reason:
Maybe, nuttx's kconfig file uses not support syntax with tools's repo's rev 891510d2534287afef6b5c90a375f94442a7e94d yet.

reference: kconfig-language's commit added some syntax.
https://gitlab.sdu.dk/sdurobotics/linux-kernels/kernel/-/commit/9059a3493efea6492451430c7e2fa0af799a2abb#541ddb1c3d900a22b15bd837aa191c0a1aa50080

SPRESENSE, 2022-01-05:
@sntulix 
Thank you for reporting the issue.

The README.md was still outdated.
Please use the `install-tools.sh` script as shown on the [Spresense SDK Getting Started Guide](https://developer.sony.com/develop/spresense/docs/sdk_set_up_en.html) to install the tools, including kconfig-frontend.
I will update README.md later.

sntulix, 2022-01-09:
@SPRESENSE 
Hi, I see.

Thank you :)

SPRESENSE, 2022-04-05:
Released SDKv2.5.0 and fixed this issue #346.

closed 2022-04-05 by SPRESENSE