sonydevworld/spresense #347: Can't get code to compile on Mac M1

cku239, 2022-01-16:
I have been following the instructions on the sony spresense developer pages to install SDK and get a basic hello world working.

The code is not compiling for me. The output has two problems maybe related?

I am using Visual Studio code on an Apple M1 mac.

When I create the workspace I get the following problem reported
"1/16/2022, 7:05:03 PM] Unable to resolve configuration with compilerPath "/arm-none-eabi-gcc".  Using "/usr/bin/clang" instead.
[1/16/2022, 7:05:03 PM] Unable to resolve configuration with compilerPath "/arm-none-eabi-gcc".  Using "/usr/bin/clang" instead.
[1/16/2022, 7:05:06 PM] Unable to resolve configuration with compilerPath "/arm-none-eabi-gcc".  Using "/usr/bin/clang" instead.
[1/16/2022, 7:05:06 PM] Unable to resolve configuration with compilerPath "/arm-none-eabi-gcc".  Using "/usr/bin/clang" instead."

I've run the cmd **arm-none-eabi-gcc -v** and this reports no errors.
This is my compilerpath setting
"compilerPath": "${config:spresense.env.toolchain.path}/arm-none-eabi-gcc",

Looking in the c_cpp_properties file I also have the following error
""${workspaceFolder}/.vscode/include"," is highlighted with a message saying it doesn't exist which it doesn't.

my workspace configuratiion is 
_"includePath": [
                "${workspaceFolder}/.vscode/include",
                "${mySpresenseSdkPath}/sdk/apps/include",
                "${mySpresenseSdkPath}/sdk/modules/include",
                "${mySpresenseSdkPath}/sdk/modules/asmp/worker",
                "${mySpresenseSdkPath}/sdk/system/include",
                "${mySpresenseSdkPath}/sdk/bsp/include",
                "${mySpresenseSdkPath}/nuttx/include",
                "${workspaceFolder}/**"
            ],_

All the other settings in the config files i.e workspace config seem to be ok.

I tried on Linux and although didnt get the warning about gcc/CLang still got it complaining about missing .vscode/include

SPRESENSE, 2022-01-19:
@cku239 
Thank you for reporting the issue.

These error messages about `compilerPath` and `.vscode/include` in the c_cpp_properties file are both known issues.
The `compilerPath` error message should be removed when you save your workspace.
And, after you run `Spresense: SDK Config`, the `.vscode/include` directory should be created and its error messeage should be removed.

Both error messages are temporary and can be ignored, but are you encountering build errors?
We have been able to use it in the Apple M1 Mac environment without any problems.

If you are encountering any config or build error, please provide detailed instructions to reproduce the problem.
Which version, VSCode, C/C++ extension, Spresense VSCode IDE and Spresense SDK, are you using? 
Please let us know the detailed procedure and what error message do you get.

Thanks,

SPRESENSE, 2022-02-24:
@cku239 As I commented before, I can't reproduce your situation. Do you have any updates?

SPRESENSE, 2022-04-05:
The original issue is not reproduced. And for additional information, resolved these error messages about `compilerPath` and `.vscode/include` in the c_cpp_properties file in the new Spresense VSCode IDE v1.3.200 version.
https://github.com/sonydevworld/spresense-vscode-ide/releases/tag/v1.3.200

Close this issue #347. Please report any problems.

Thanks.

closed 2022-04-05 by SPRESENSE