sonydevworld/spresense #426: Issue with SDK getting stated guide

MicroTensor, 2022-04-24:
Hi,

Today for the first time I was following SDK getting stated guide, and as per instructions at https://developer.sony.com/develop/spresense/docs/sdk_set_up_en.html#_configuration_instructions
should be running `tools/config.py examples/hello` but the latest SDK has layout of `apps/examples/hello`
and when I ran `tools/config.py apps/examples/hello` get error 
[image]

SPRESENSE, 2022-04-25:
Hi,

The arguments of the `tools/config.py` are documented correctly.
```
$ cd spresense/sdk
$ tools/config.py examples/hello
```
The `examples/hello` argument specifies the configuration name under `spresense/sdk/configs`, which is not the path to the source code. And the available arguments can be listed in `tools/config.py -l`.

Thanks.

KazuyaHioki, 2022-07-26:
Close #426

closed 2022-07-26 by KazuyaHioki