sonydevworld/spresense #456: build_env.sh source error

emi-tsato, 2022-07-21:
I followed the Spresense SDK `Getting Started Guide` and successfully finished. But failed to do `Examples and Tutorials` due to the following error.

```
% source ~/spresenseenv/setup <= this did not make any difference
% source tools/build-env.sh 
tools/build-env.sh:225: = not found
```

My environment is Mac (macOS 12.4, M1 Pro). 

The line 225 is the following code.

https://github.com/sonydevworld/spresense/blob/master/sdk/tools/build-env.sh#L225

When I simply changed the equation to the one with `-z`, this will pass. But encountered another `= not found error` as below.

```
% source tools/build-env.sh
Warning: Spresense user application directory is not set.
         Please run
         $ spr-set-approot <application home directory>
=======================================
cat: /Users/takenori/Documents/sdk/tools/mkversion.sh: No such file or directory
   SDK_VERSION = 
cat: /Users/takenori/Documents/sdk/tools/mkversion.sh: No such file or directory
 NUTTX_VERSION = 
 SPRESENSE_SDK = /Users/takenori/Documents
SPRESENSE_HOME = 
   GCC_VERSION = arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
          HOST = Darwin arm64
=======================================
Info: Please update the version to 4.0 or later
      The spresense tools completion can not work well
      because of old bash version.
% spr-set-approot /Users/takenori/Documents/open_repos/spresense 
spr-set-approot:4: = not found
% spr-create-approot ~/Documents/open_repos/my_sp_apps
spr-create-approot:4: = not found
```

It looks this is caused simply because `==` is used for the comparison. Is there anything I missed here?

SPRESENSE, 2022-07-22:
Hi,

`build-env.sh` must be run on bash shell. Please check the shell of your execution environment.
```
$ echo $SHELL
/bin/bash
```
If it is not `bash`, switch to `bash` and then run it.
```
% bash
$ source tools/build-env.sh
```

emi-tsato, 2022-07-22:
Thanks, it did work!

closed 2022-07-22 by emi-tsato