sonydevworld/spresense #493: Why should we enable power_button pin of alt1250?

takumiando, 2022-10-07:
Could you give me more details about this problem and its fix?

```
commit df57c1cba4b67642c75ad57a1ff722e6b0ddd924
Author: SPRESENSE <41312067+SPRESENSE@users.noreply.github.com>
Date:   Thu Jul 28 19:09:25 2022 +0900

    drivers/modem/alt1250: Temporary enable power_button pin
    
    Some LTE board cannot boot up correctly right now.
    With this fix, problem LTE board can boot up correctly.
    So, apply this fix before apply main fix.

diff --git a/boards/arm/cxd56xx/spresense/src/cxd56_alt1250_power.c b/boards/arm/cxd56xx/spresense/src/cxd56_alt1250_power.c
index 4f19b26921..24fa3cf054 100644
--- a/boards/arm/cxd56xx/spresense/src/cxd56_alt1250_power.c
+++ b/boards/arm/cxd56xx/spresense/src/cxd56_alt1250_power.c
@@ -81,6 +81,10 @@ void board_alt1250_poweron(void)
 
   cxd56_gpio_write(ALT1250_SHUTDOWN, true);
 
+  /* Workaround: Power button pin set to high before power on */
+
+  cxd56_gpio_write(ALT1250_LTE_POWER_BUTTON, true);
+
   board_power_control(POWER_LTE, true);
 }
```

takumiando, 2022-10-07:
I couldn't find the issue form of https://github.com/sonydevworld/spresense-nuttx.
So I posted here.

SPRESENSE, 2022-10-18:
@takumiando san,
Due to component variation, some LTE boards did not turn on with the past power sequence.
With this change, we are revising the sequence to reduce component variation.

takumiando, 2022-10-20:
@SPRESENSE 
I understand. Thank you.