sonydevworld/spresense #510: gs2200m association failed : retrying : multiwebcam

RenaultCorse, 2022-11-23:
I have a problem about WiFi Connection in running sample named multiwebcam.
I used gs2200m command for setting WiFi connection to our AccessPoint.
But sometimes, "association failed : retrying" error happened after gs2200m command.

The following link is Screen Shot.
[https://www.dropbox.com/t/cv5zoAwnLDIGtGFR](url)

Please give me any idea.
Thanks

RenaultCorse, 2022-11-23:
And init.rc is as follows.
`gs2200m SSID WiFiPass &
sleep 5
ifconfig &
multiwebcam &`

takayoshi-k, 2023-01-10:
Hi @RenaultCorse 
This message is displayed when an attempt to connect to the access point fails and times out.
If you have already made sure that your SSID and password are correct, it is possible that the encryption method requested by your access point is not supported by the GS2200M.
Please check first to see if you can connect to another access point using WiFi tethering on your phone, for example.