sonydevworld/spresense #537: Unable to setup SDK CLI for windows 10

Hrithik2212, 2023-01-13:
I have been following the getting started guide for SDK CLI for windows 10 from the the website [https://developer.sony.com/develop/spresense/docs/sdk_set_up_en.html#_development_environment](url) 
After installing MYSYS2 . I opened MYSYS2 MYSYS and tried the following command 

    curl -L https://raw.githubusercontent.com/sonydevworld/spresense/master/install-tools.sh > install-tools.sh
I got an error stating it was unable to connect to the port 
Thus I manually copy pasted the content from the github Repository file 
Then when I ran this cmd 

    bash install-tools.sh

I got the message 
Sorry this platform is not supported 
Even though I had installed MYSYS2 and had a windows 10 OS 

Thus I changed the following code in the install-tools.sh file from line 330 

     case "`uname -s`" in
    Linux)
        OS=linux
        ;;
    MSYS2*)
        OS=win
        ;;
    Darwin)
        OS=mac
        ;;
    *)
        echo Sorry, this platform is not supported.
        exit 1
        ;;
    esac

to 

    case "`uname -s`" in
    Linux)
        OS=linux
        ;;
    MSYS2*)
        OS=win
        ;;
    Darwin)
        OS=mac
        ;;
    *)
        OS=win
        ;;
        # echo Sorry, this platform is not supported.
        # exit 1
        # ;;
    esac

With this ,the problem was solved and all the packages got installed , however I am not sure if I everything is working properly . This is only a temporary solution for windows 10 users and I request repository owners to look into it . Thank you

SPRESENSE, 2023-01-16:
Hi @Hrithik2212 

I think the error of `curl` is due to your network not working properly.
Running `curl` with the `-v` option may give you any hints.

Also, do you make a mistake in the description of `uname` due to manual copying? The correct is `MSYS*)`, not `MSYS2*)`.
If you're still having problems, please let me know the result of `uname -s`.

Finally, I suspect that the missing `spresenseenv/setup` file is also a network problem or manual copy error that is causing the installation to fail. Are there any errors while running the `install-tools.sh` script? Please check for your network environment.

Hrithik2212, 2023-01-21:
Hi  @SPRESENSE , thank you for the feedback . I tried the curl command -v option and it worked , however the   

     spresenseenv/setup    
file is not  available 
This is the output when i ran 

      bash install-tools.sh

_*Output*_

    == Install base command line tools
    there is nothing to do
    == Install additional tools
    == Install cross toolchain
    install-tools.sh: line 393: ${_setup}: ambiguous redirect
    install-tools.sh: line 394: ${_setup}: ambiguous redirect
    install-tools.sh: line 395: ${_setup}: ambiguous redirect
    Installation is done.

SPRESENSE, 2023-01-23:
Hi @Hrithik2212, 
The error may be due to a blank space in your HOME directory. Does your username contain spaces?

Try modifying and running the script as follows.

```diff
diff --git a/install-tools.sh b/install-tools.sh
index 5c23575..92bf5c4 100755
--- a/install-tools.sh
+++ b/install-tools.sh
@@ -436,10 +436,10 @@ fi
 # Create PATH environment setup support script

 _setup=${HOME}/${SPRBASENAME}/setup
-mkdir -p ${HOME}/${SPRBASENAME}
+mkdir -p "${HOME}/${SPRBASENAME}"

-echo "#!/usr/bin/env bash" > ${_setup}
-echo "PATH=${SPRROOT}/usr/bin:"'${PATH}' >> ${_setup}
-echo "export PATH" >> ${_setup}
+echo "#!/usr/bin/env bash" > "${_setup}"
+echo "PATH=${SPRROOT}/usr/bin:"'${PATH}' >> "${_setup}"
+echo "export PATH" >> "${_setup}"

 echo "Installation is done."
```

By the way, Spresense SDK has a restriction that directory names containing spaces cannot be used.
https://developer.sony.com/develop/spresense/docs/sdk_set_up_en.html#_spresense_sdk_may_not_be_used_depending_on_the_folder_namepath_used

When you clone the Spresense SDK from GitHub, clone it to a path that does not contain spaces.

Thanks.

Hrithik2212, 2023-01-29:
@SPRESENSE  Thank you for your feedback

Yes my home directory did have a space in between and I have resolved the issues by changing my path variable in etc/passwd file 

Now I am facing issues in flashing the board as it seems I don't have the screen command in my windows 10 laptop, I am unable to find any resources related to it on the internet 

I got this message when I 

     $ screen /dev/ttyUSB 115200
     -bash: screen: command not found

I don't know how to install screen or maybe its not in my path environment and I don't which folder to set in the path environment

SPRESENSE, 2023-01-30:
@Hrithik2212 
The `screen` command is a procedure for Linux and macOS, so please use a serial terminal (`PuTTY`, `TeraTerm`, etc.) that can be used on Windows.
Thanks.

Hrithik2212, 2023-01-30:
@SPRESENSE 
I did install PuTTY on my windows 10 laptop , it works when I use the PuTTY cmd on my command prompt , however it doesn't work when I use it in MYSYS terminal which shows 

    -bash: PuTTY: command not found

When I run just PuTTY on cmd or on bash on windows powershell a GUI PuTTY configuration windows open

SPRESENSE, 2023-01-31:
@Hrithik2212 
The reason you can't run putty from the MSYS2 terminal is because your `PATH` is not set.
```
$ export PATH=<your path to PuTTY>:$PATH
e.g.)
$ export PATH=/c/Program\ Files/PuTTY:$PATH
```

You can also take over the Windows environment variable `PATH` by setting `MSYS2_PATH_TYPE=inherit`, as shown in `msys2_shell.cmd` in your MSYS installation location.
```
rem To export full current PATH from environment into MSYS2 use '-use-full-path' parameter
rem or uncomment next line
rem set MSYS2_PATH_TYPE=inherit
````
It may work if you set the Windows environment variable `MSYS2_PATH_TYPE` to `inherit`.
See the MSYS2 documentation for details.

And, when using putty from the command line, you must specify the options.
```
$ putty -serial <your COM port number> -sercfg 115200
e.g.)
$ putty -serial COM12 -sercfg 115200
```
See the PuTTY documentation for more information.
https://documentation.help/PuTTY/using-cmdline.html

Hrithik2212, 2023-03-15:
Thank you , I was finally able to setup the sdk

closed 2023-03-15 by Hrithik2212