sonydevworld/spresense #625: The SDK example fsperf not working.

dlukian, 2023-03-30:
I tried loading the example fsperf and after successfully flashing the nuttx.spk, the Spresense board cannot be communicated to via the serial connection. I attempted to communicate via the Spresense SDK VSCode IDE "Open Serial Port" - the terminal remains blank. I tried using the `$screen /dev/ttyUSB0 115200` - screen remains blank.

I can flash another example successfully and able to then connect to the board via serial port.

Using the latest 3.0 bootloader version.

SPRESENSE, 2023-03-30:
Hi @dlukian san

Thank you for reporting the issue. This `fsperf` configuration was assuming that an `eMMC Add-on board`  was connected to the Add-on connector or nothing was connected. Is there another board connected to your Add-on connector?

Could you please remove the following two lines from `spresense/sdk/configs/examples/fsperf/defconfig` and try it.
```
+CXD56_EMMC=y
+CXD56_EMMC_POWER_PIN_I2S0_BCK=y
```
Thanks,

dlukian, 2023-03-30:
I followed your instructions and `fsperf` now works as described. Thank you for prompt response.

I have an add-on board with SD card slot. I am using the PDA01 board.

closed 2023-03-30 by dlukian