sonydevworld/spresense #748: Issue with the tools/build-env.sh file

Abdullah-Shahid01, 2024-08-02:
Hi, 

I am trying to go through the build process on Sony Spresense, but whenever I run the command source tools/build-env.sh I get an error saying the following:
tools/build-env.sh:225: = not found

SPRESENSE, 2024-08-05:
Hi, @Abdullah-Shahid01, 

Is there a problem with your bash path settings?
When you enter "bash --version", does the version number appear?

Also, Is your build platform environment one of the following?
https://developer.sony.com/spresense/development-guides/sdk_set_up_en.html#_development_environment

Abdullah-Shahid01, 2024-08-07:
Output of "bash --version"
[image]

Output of "arm-none-eabi-gcc -v"
gcc version 10.3.1 20210824 (release) (GNU Arm Embedded Toolchain 10.3-2021.10)

I am not sure if this is relevant but I am using an Apple M1 macbook.

SPRESENSE, 2024-08-08:
Hi @Abdullah-Shahid01 

On the Apple M1 macbook, the default shell is `zsh` and it's not possible to run `build-env.sh` on `zsh` shell.

Please switch to `bash` shell and then run `source tools/build-env.sh` on your terminal.
```
$ bash
$ source tools/build-env.sh
```