sonydevworld/spresense #826: Loader Version Mismatch Error

GOROman, 2025-03-01:
## Loader Version Mismatch Error

### Problem
When executing the tools/flash.sh script, a loader version mismatch error occurs. According to the error message, loader version v3.4.0 is required, but the download link points to v3.3.0.

### Steps to Reproduce
1. Execute `tools/config.py examples/hello`
2. Download v3.3.0 from the provided link in the warning message
3. Run `./tools/flash.sh -e ~/Downloads/spresense-binaries-v3.3.0.zip`

### Error Message
```zsh
$ ./tools/config.py examples/hello
WARNING: New loader v3.4.0 is required, please download and install.
         Download URL   : https://developer.sony.com/file/download/download-spresense-firmware-v3-3-0
         Install command:
                         1. Extract loader archive into host PC.
                            ./tools/flash.sh -e <download zip file>
                         2. Flash loader into Board.
                            ./tools/flash.sh -l /Users/goroman/Work/github/spresense/firmware/spresense -c <port>
```

```zsh
$ ./tools/flash.sh -e ~/Downloads/spresense-binaries-v3.3.0.zip

Error: Please use correct loader version (Selected: v3.3.0, Required: v3.4.0).
```

### Environment
- OS: macOS
- Spresense SDK version: v3.4.0 (b308223fe058bb0a91887df51f6e2aa76f13e22d)
- Downloaded binary: spresense-binaries-v3.3.0.zip https://developer.sony.com/ja/spresense/downloads/firmware-binary

### Expected Behavior
The download URL should point to the correct v3.4.0 version, or the warning message should provide appropriate information about the available correct version.

### Additional Information
The warning message indicates that loader v3.4.0 is required, yet the download URL points to v3.3.0 binaries, causing the mismatch.

GOROman, 2025-03-01:
**Workaround:** 

I was able to proceed by installing Arduino IDE v3.4.0 and using ```firmware.zip```.

[image]

[image]

SPRESENSE, 2025-03-03:
Thank you for reporting the issue.
We have released a corrected version as SDKv3.4.0.
There is no firmware update this time.
And we have updated the [releasenote](https://github.com/sonydevworld/spresense/releases/tag/v3.4.0).

The SDK tags have been updated, so if you have already downloaded them, you will need to force update the tags with the following command:

```
git fetch --tags --force
```

Thank you for your cooperation.

GOROman, 2025-03-03:
I understand that there are no changes to the firmware itself. Thank you.

I will close this issue.

closed 2025-03-03 by GOROman