sonydevworld/spresense #830: Broken Links in IMU Example Section (404 - page not found)

GOROman, 2025-03-04:
## Overview
The links to the IMU example and IMU RAW data logging example in the Spresense developer documentation are broken, leading to 404 Not Found errors.

## Details
- **Documentation URL**: [Spresense SDK Developer Guide](https://developer.sony.com/spresense/development-guides/sdk_developer_guide_ja.html#imu_sample_codes)

[image]

- **Broken Links**:
  - **IMU example**: [Current link](https://github.com/sonydevworld/spresense/tree/master/examples/cxd5602pwbimu/cxd5602pwbimu/) → **404 - page not found**
  - **IMU RAW data logging example**: [Current link](https://github.com/sonydevworld/spresense/tree/master/examples/cxd5602pwbimu/cxd5602pwbimu_logger/) → **404 - page not found**

## Suggested Fix
The correct links should be:
- **IMU example**: [Correct link](https://github.com/sonydevworld/spresense/tree/master/examples/cxd5602pwbimu)
- **IMU RAW data logging example**: [Correct link](https://github.com/sonydevworld/spresense/tree/master/examples/cxd5602pwbimu_logger)

## Screenshots
**Please see attached screenshots of the 404 errors.**

[image]

SPRESENSE, 2025-03-05:
Sorry it was accidentally closed, and reopen.
The corrected documents will be published soon.

SPRESENSE, 2025-03-05:
@GOROman 
Thank you for your kind issue report.
The correct links are as you indicated.
I have now published the fixed documents.

closed 2025-03-09 by GOROman