sonydevworld/spresense #832: 3.4.0 invalid bootloader warning message

nakajimayoshi, 2025-03-08:
When I go to run tools/config.py I get a warning message

```
WARNING: New loader v3.4.0 is required, please download and install.
         Download URL   : https://developer.sony.com/file/download/download-spresense-firmware-v3-3-0
         Install command:
                          1. Extract loader archive into host PC.
                             ./tools/flash.sh -e <download zip file>
                          2. Flash loader into Board.
                             ./tools/flash.sh -l /Users/nakajimayoshi/Documents/projects/embeded/spresense/firmware/spresense -c <port>
```

However there's a few issues. 

1. The message says `new loader v3.4.0`, however the link points to a download for firmware `v3-3-0` 

2. Manually entering the address: `https://developer.sony.com/file/download/download-spresense-firmware-v3-4-0` points to a missing URL. Does the 3.4.0 firmware simply not exist?

3. On macOS, these files are not downloaded as a .zip, they are downloaded as a normal file. 

4. After zipping the the folder, running the command 

```bash
./tools/flash.sh -e /Users/nakajimayoshi/Downloads/spresense-binaries-v3.zip
```

results in the following message:
`ERROR: Please select correct zip file`

How am I supposed to get rid of this warning message then?

SPRESENSE, 2025-03-10:
Thank you for your report.
We have released a corrected version as SDKv3.4.0 to address the issue #826.
There is no firmware update this time.

##  The Loader version message for SDK v3.4.0 on macOS
[image]

## Force fetch for tag
The SDK tags have been updated, so if you have already downloaded them, 
you will need to force update the tags with the following command:

```
git fetch --tags --force
```

Thank you for your cooperation.

GOROman, 2025-03-13:
It is my opinion that the versions provided on the Arduino IDE side are not consistent with v3.4.0 and should be unified with v3.4.0. Users will be confused.

nakajimayoshi, 2025-03-14:
I reinstalled to repository to fix the problem but it re-appeared as 3.4.1, which also doesn't work. And has lead to a further problem. See #839

nakajimayoshi, 2025-03-27:
@SPRESENSE 

Basically step 1 as written in the warning message leads to theerror. 
What I had to do instead was download, and copy the unzipped (not zipped) files to /spresense/firmware/spresense.

Then, run the command in step 2 which finally worked. 

I think it's mostly a problem of unclear instructions.