sonydevworld/spresense #839: Can't compile hello world program

nakajimayoshi, 2025-03-14:
Following a reinstall of Spresense SDK following the [tutorial](https://developer.sony.com/spresense/development-guides/sdk_set_up_ide_en.html) exactly as written for macOS I can no longer compile a basic program for the Spresense board using visual studio or the command line.

Executing the visual studio application build command or using the cli results in the following error

```
Register: sh
./stdlib/lib_strtold.c:44:10: fatal error: math.h: No such file or directory
   44 | #include <math.h>
      |          ^~~~~~~~
compilation terminated.
ERROR: arm-none-eabi-gcc failed: 1
       command: arm-none-eabi-gcc -MT bin/lib_strtold.o  -M '-Wstrict-prototypes' '-Wno-attributes' '-Wno-unknown-pragmas' '-Wno-psabi' '-mabi=aapcs' '-Os' '-fno-strict-aliasing' '-fomit-frame-pointer' '-fno-common' '-Wall' '-Wshadow' '-Wundef' '-ffunction-sections' '-fdata-sections' '-g' '-mlittle-endian' '-march=armv7e-m' '-mtune=cortex-m4' '-mfpu=fpv4-sp-d16' '-mfloat-abi=hard' '-mthumb' '-Wa,-mthumb' '-Wa,-mimplicit-it=always' '-isystem' '/Users/jacobsullivan/Documents/sandbox/sony/spresense/nuttx/include' '-I.' '-isystem' '/Users/jacobsullivan/Documents/sandbox/sony/spresense/nuttx/../sdk/include' '-D__NuttX__' '-DNDEBUG' '-D__KERNEL__' '-pipe' '-I' '/Users/jacobsullivan/Documents/sandbox/sony/spresense/nuttx/libs/libc' ./stdlib/lib_strtold.c
make[3]: *** [lib_strtold.ddc] Error 1
make[2]: *** [.depend] Error 2
make[1]: *** [pass2dep] Error 2
make: *** [all] Error 2

 *  The terminal process "/bin/zsh '-l', '-c', '.vscode/build.sh build'" terminated with exit code: 2. 
 *  Terminal will be reused by tasks, press any key to close it. 
``` 

This is the source code of the program

```c
#include <nuttx/config.h>
#include <stdio.h>

int main(int argc, char *argv[])
{
  return 0;
}
```

SPRESENSE, 2025-03-18:
Judging from the error, it seems that the environment is not as we expected.
What happens if you build the Spresense SDK code referenced on your vscode environment
using a CLI command on the terminal? 
For example, in the Hello sample, change the directory to spresense/sdk on the terminal 
and try building the hello example to see the results.

Example command:
```
$make clean
$tools/config.py examples/hello
$make
```
If successful, nuttx.spk will be generated.

If no error occurs here,
Please try building again in vscode by reselecting the configuration in the step: [sdk_configuration](https://developer.sony.com/spresense/development-guides/sdk_set_up_ide_en.html#_sdk_configuration) . 
If necessary, clean the code before selecting the configuration.

If any error occurs on the CLI,
Please try starting over from setup at the step: [setup_for_macos](https://developer.sony.com/spresense/development-guides/sdk_set_up_ide_en.html#_setup_for_macos) and then try building following the above [sdk_configuration](https://developer.sony.com/spresense/development-guides/sdk_set_up_ide_en.html#_sdk_configuration) step.

nakajimayoshi, 2025-03-18:
re running the config and clean seemed to fix the issue, thanks
```
sdk nakajimayoshi$ make clean
sdk nakajimayoshi$ tools/config.py examples/hello
WARNING: New loader v3.4.1 is required, please download and install.
         Download URL   : https://developer.sony.com/file/download/download-spresense-firmware-v3-4-1
         Install command:
                          1. Extract loader archive into host PC.
                             ./tools/flash.sh -e <download zip file>
                          2. Flash loader into Board.
                             ./tools/flash.sh -l /Users/nakajimayoshi/Documents/sandbox/sony/spresense/firmware/spresense -c <port>
sdk nakajimayoshi$ make
Create version.h
LN: platform/board to /Users/nakajimayoshi/Documents/sandbox/sony/spresense/sdk/apps/platform/dummy
Register: hello
Register: nsh
Register: sh
CPP:  /Users/nakajimayoshi/Documents/sandbox/sony/spresense/nuttx/boards/arm/cxd56xx/spresense/scripts/ramconfig-new.ld-> /Users/nakajimayoshi/Documents/sandbox/sony/spresense/nuttx/boards/arm/cxd56xx/spresenseLD: nuttx
Generating: nuttx.spk
tools/cxd56/mkspk -c2 nuttx nuttx nuttx.spk;
File nuttx.spk is successfully created.
Done.
```